Autonomy-Logic/openplc-editor #248: Change main POU name

AndBat86, 2025-05-15:
## Expected Behavior

I should create a new POU (PrgMain for example) and set it inside Resurces to be my main POU.

## Current Behavior

The error  "Main POU not found. The project must have a "main" program pou." appears during build process.

## Steps to Reproduce

## Possible Solution (Not obligatory)

## Context

## Your Environment

- Version : 4.0.3_beta
- Operating System and version : appimage

thiagoralves, 2025-05-16:
The compiler requires that you have at least one POU named "main". You can create additional POUs named something else and add them to additional tasks, but without a "main" POU, your program won't compile. Do you have a requirement to rename the "main" POU to "PrgMain"?

AndBat86, 2025-05-16:
I definitely don't need to rename the POU. "PrgMain" is only the name convention that I use in Codesys. But, probably, it is better to lock ability to edit (or delete, since I wasn't able to edit any POU name) the "main" POU if it is mandatory. Thanks a lot.

thiagoralves, 2025-11-01:
Fixed in the development branch