Autonomy-Logic/openplc-editor #280: fails to build on linux

dvc94ch, 2025-07-02:
## Expected Behavior

`npm install` succeeds

## Current Behavior

```
npm install
> open-plc-editor@4.0.4-beta postinstall
> ts-node scripts/check-native-dep.js && electron-builder install-app-deps && npm run build:dll

  • electron-builder  version=25.1.8
  • loaded configuration  file=/home/dvc/openplc-editor/electron-builder.json
  • installing production dependencies  platform=linux arch=x64 appDir=/home/dvc/openplc-editor/release/app
  • executing @electron/rebuild  electronVersion=35.0.0 arch=x64 buildFromSource=false appDir=release/app
  • installing native dependencies  arch=x64
  • completed installing native dependencies

> open-plc-editor@4.0.4-beta build:dll
> cross-env NODE_ENV=development TS_NODE_TRANSPILE_ONLY=true webpack --config ./configs/webpack/webpack.config.renderer.dev.dll.ts

[webpack-cli] Failed to load '/home/dvc/openplc-editor/configs/webpack/webpack.config.renderer.dev.dll.ts' config
[webpack-cli] TypeError [ERR_IMPORT_ATTRIBUTE_MISSING]: Module "file:///home/dvc/openplc-editor/release/app/package.json" needs an import attribute of "type: json"
    at validateAttributes (node:internal/modules/esm/assert:88:15)
    at defaultLoadSync (node:internal/modules/esm/load:178:3)
    at #loadAndMaybeBlockOnLoaderThread (node:internal/modules/esm/loader:836:12)
    at #loadSync (node:internal/modules/esm/loader:858:49)
    at ModuleLoader.getModuleJobForRequire (node:internal/modules/esm/loader:488:38)
    at #link (node:internal/modules/esm/module_job:429:34)
    at new ModuleJobSync (node:internal/modules/esm/module_job:401:17)
    at ModuleLoader.getModuleJobForRequire (node:internal/modules/esm/loader:522:11)
    at #link (node:internal/modules/esm/module_job:429:34)
    at new ModuleJobSync (node:internal/modules/esm/module_job:401:17) {
  code: 'ERR_IMPORT_ATTRIBUTE_MISSING'
}
npm error code 2
npm error path /home/dvc/openplc-editor
npm error command failed
npm error command sh -c ts-node scripts/check-native-dep.js && electron-builder install-app-deps && npm run build:dll
npm error A complete log of this run can be found in: /home/dvc/.npm/_logs/2025-07-02T10_31_23_059Z-debug-0.log
```

## Steps to Reproduce

Apply this diff to fix previous build errors:

```diff
diff --git a/configs/webpack/webpack.config.renderer.dev.dll.ts b/configs/webpack/webpack.config.renderer.dev.dll.ts
index f5ef3b54..aac36c35 100644
--- a/configs/webpack/webpack.config.renderer.dev.dll.ts
+++ b/configs/webpack/webpack.config.renderer.dev.dll.ts
@@ -7,9 +7,9 @@ import { join } from 'path'
 import webpack from 'webpack'
 import { merge } from 'webpack-merge'

-import { dependencies } from '../../package.json'
-import checkNodeEnv from '../../scripts/check-node-env'
-import baseConfig from './webpack.config.base'
+import { dependencies } from '../../package.json' with { type: "json" }
+import checkNodeEnv from '../../scripts/check-node-env.js'
+import baseConfig from './webpack.config.base.ts'
 import webpackPaths from './webpack.paths'
```

## Possible Solution (Not obligatory)

Probably development is done using some old node version?

## Context

Trying to build the editor.

## Your Environment

- Version : development branch
- Operating System and version : `Linux xps13-9320 6.15.3-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 19 Jun 2025 14:41:19 +0000 x86_64 GNU/Linux`

JoaoGSP, 2025-07-02:
Hi, @dvc94ch !
Could you please share what is the Node version that you're using? And the content of your `release/app/package.json` file?

dvc94ch, 2025-07-02:
ah, that fixed the issue thx. don't do much/any js development. now got this error:

```
npm run start:dev

> open-plc-editor@4.0.4-beta start:dev
> ts-node scripts/check-port-in-use.js && npm run prestart && npm run start:renderer

> open-plc-editor@4.0.4-beta prestart
> cross-env NODE_ENV=development TS_NODE_TRANSPILE_ONLY=true webpack --config ./configs/webpack/webpack.config.main.dev.ts

ERROR in /home/dvc/openplc-editor/src/renderer/screens/workspace-screen.tsx
163:72-88
[tsl] ERROR in /home/dvc/openplc-editor/src/renderer/screens/workspace-screen.tsx(163,73)
      TS2322: Type '"Pin Mapping" | "Configuration"' is not assignable to type '"Configuration"'.
  Type '"Pin Mapping"' is not assignable to type '"Configuration"'.

webpack compiled with 1 error
```

JoaoGSP, 2025-07-02:
I just spot this error rsrsrs... I pushed a fix in dev branch right now, just pull to your local and we should be good to go 😄 .