Autonomy-Logic/openplc-editor #294: #include <P1AM.h> - include file not found and timer_us not declared in scope (p1am.c)

Dowster64, 2025-07-30:
## Expected Behavior

transfer to PLC works

## Current Behavior

When trying to transfer to PLC board type P1AM-100 [1.8.14]  an error during build occurs:
```
In file included from C:\Users\admin\OpenPLC_Editor\editor\arduino\examples\Baremetal\arduino_libs.h:13:0,
                 from C:\Users\admin\OpenPLC_Editor\editor\arduino\examples\Baremetal\Baremetal.ino:22:
C:\Users\admin\OpenPLC_Editor\editor\arduino\examples\Baremetal\modules\p1am.c: In function 'uint8_t p1am_init()':
C:\Users\admin\OpenPLC_Editor\editor\arduino\examples\Baremetal\modules\p1am.c:15:9: error: 'timer_us' was not declared in this scope
         timer_us = micros() + scan_cycle;
         ^~~~~~~~
C:\Users\admin\OpenPLC_Editor\editor\arduino\examples\Baremetal\modules\p1am.c:15:9: note: suggested alternative: 'timer_t'
         timer_us = micros() + scan_cycle;
         ^~~~~~~~
         timer_t
```

## Steps to Reproduce

1.  Open Blink_P1AM example

2. No changes necessary - just transfer program to PLC

3. Same error

4.

## Possible Solution (Not obligatory)

## Context

Trying to use OpenPLC in  a realworld example to replace an old SAIA PCA (PLC).  Linux runtime works well.  Using FUXA SCADA and modbus interface works well

## Your Environment

- Version : OpenPLC release 2025-06-09 (using venv)
- Operating System and version : Windows 11 Pro. 24H2

thiagoralves, 2025-07-30:
Fixed! Thanks for reporting this. Update your editor to the latest version (2025-07-29) and let me know if it works for you

closed 2025-07-30 by thiagoralves

Dowster64, 2025-07-30:
I'll check this out - thanks for being awake / in TZ !  I added
//SJD test
uint32_t timer_us = 0;
in p1am.c and got it to compile...  I'l see what the real fix is!

thiagoralves, 2025-07-30:
That's what I was about to write when I read your first message. Most likely your P1AM was in a bad state, and needed to be placed in bootloader mode. The P1AM needs a working firmware to be able to receive new programs. If its firmware has crashed or is corrupted, the only way out is placing it in bootloader mode. 

Now that you have uploaded a working program, you should be able to upload the next one normally, without having to enter in bootloader mode.