Autonomy-Logic/openplc-editor #296: P1AM-200 has the same limited modbus range for IO as the P1AM-100, it should be much larger?

Dowster64, 2025-08-03:
Hi,
I have tried using an output coil %QX7.0 on the Blink_P1AM example with the  P1AM-200 [1.7.13] board selected instead of the 100 series.   It still has the same limit as the 100 series and compilation fails:
```
$ C:\Users\xxxxx\OpenPLC_Editor\editor\arduino\bin\iec2c.exe -v
matiec version 0.1
changeset id: 
$? = 0
$ C:\Users\xxxx\OpenPLC_Editor\editor\arduino\bin\arduino-cli-w64.exe --no-color version
arduino-cli  Version: 1.0.3 Commit: 8b6ad258 Date: 2024-07-23T08:45:25Z
$? = 0
Checking Core and Board installation...
Core FACTS:samd is installed

Checking required libraries...
All required libraries are already installed.
Compiling .st file...
$ C:\Users\xxxxxx\OpenPLC_Editor\editor\arduino\bin\iec2c.exe -f -l -p plc_prog.st
POUS.c
POUS.h
LOCATED_VARIABLES.h
VARIABLES.csv
Config0.c
Config0.h
Res0.c
$? = 0
Copying HAL source file...
Adding arduino sketch file C:\Users\xxxxx\OpenPLC_Editor\editor\arduino\examples\Baremetal\ext\arduino_sketch.h...
Generating definitions file 'C:\Users\xxxxx\OpenPLC_Editor\editor\arduino\examples\Baremetal\defines.h'...
Error: wrong location for var __QX7_0
Build phase <function build.<locals>.generate_glue_code at 0x0000018E88E78D30> failed
```

Is it possible to configure this and other modbus addresses to a much larger value?

Thanks,

thiagoralves, 2025-08-03:
For general questions, please use the forum at https://openplc.discussion.community/

Dowster64, 2025-08-05:
@thiagoralves , I would have thought this was a feature enhancement as it might be controlled by openplc.h MAX_DIGITAL_INPUT and MAX_DIGITAL_OUTPUT? (or at least I guess it is)

thiagoralves, 2025-08-05:
Ah, got it. Yeah, this can be changed from the OpenPLC side. I'll leave it open as enhancement then. Thanks.

JoaoGSP, 2026-10-20:
This issue was resolved on the https://github.com/Autonomy-Logic/openplc-editor/pull/1093 PR, will be available in the next editor release.

closed 2026-10-20 by JoaoGSP