Autonomy-Logic/openplc-editor #322: Console event timestamps are not static

mdkrieg, 2025-09-13:
## Expected Behavior

Editor console (the one in the application) timestamps of previously printed events should be static, representing the time the event occurred

## Current Behavior

Whenever a new event is printed, the previously printed events all update their timestamp to the time of the latest event.

## Steps to Reproduce

1. Open a project and press build

2. Keep an eye on the log, watching the top (oldest) events, you will see the timestamps change

3. You can press build again and timestamps of previous builds will also change

## Possible Solution (Not obligatory)

## Context

## Your Environment

- Version : 4.0.5-beta
- Operating System and version : Windows 10 Pro 22H2 19045.6332

JoaoGSP, 2025-09-15:
> My solution, looks to be working on my end:
> 
> [mdkrieg@523219f](https://github.com/mdkrieg/openplc-editor-mdk/commit/523219fa83933e82f5bb71c66f4c154d11650844)
> 
> I added attribute tstamp to the base log class as a date type, then am doing the conversion to string at the react component level.

Hey!
Thanks a lot for checking out OpenPLC and reporting this. We’ve already spotted the issue and it’s fixed in our current development branch. The fix will be part of the next release, which should be out really soon! 🚀

JoaoGSP, 2025-09-22:
Issue solved!